Tom-Hirschberger/MMM-GPIO-Notifications #6: Keep screen on until no more motion is detected.

Clintvd, 2022-10-16:
Hi Tom, 

First off all thank you for this guide, it's very helpful.
This is more a question than an issue.

I have installed everything correctly and the pir sensor detects my motion and turns off the screen once the (delay) time has passed. So far everything good. But what I want is that the screen remains on until no motion is detected for x amount of time. I hope you can help me with this.

Thank you in advance.

Tom-Hirschberger, 2022-10-16:
Hi,

great that you like the guide and the effort was worth it.
If i understand correctly you configured MMM-GPIO-Notifications and MMM-Screen-Powersave-Notification, right?

You want your monitor to turn on if motion was detected, keep it on as long as motion is detected (within the delay of MMM-Screen-Powersave-Notification) and turn of the screen if no more motion was detected within the delay, right?

This should be the normal be the case if you followed the guide.

Could be please post the two configuration blocks? 
Did you configure the delay within MMM-GPIO-Notifications or/and MMM-Screen-Powersave-Notification?

Clintvd, 2022-10-17:
Hi,

I have set the delay on 30 seconds ( for testing purpose). The thing is, after 30 seconds the screen goes dark and if no further movement is detected it stays dark which is good but when i'm still in front of the sensor(and moving) it goes dark and turns back on again. I could just increase the delay time but my idea was that the mirror goes on and stays on until there is no one in front of it. 

My modules below.

module: 'MMM-GPIO-Notifications',
			config: {
				'4': {
						gpio_state: 1,
						gpio_debounce: 0,
				delay: 30000,
						notifications: [
							{ 
										notification: 'SCREEN_ON', 
								payload: { 'forced': false }
							},
module: 'MMM-Screen-Powersave-Notification',
				config: {
					delay: 30,
					}\

Tom-Hirschberger, 2022-10-17:
Hi,

it looks like you configured the wrong delay. The delay of the Gpio module causes triggers of the sensor to be ignored for a specific amount of time. The delay of the Screensave module configures how long the screen stays active if no more activity was detected.
So you simply need to set the delay of GPIO to a lower value (say 5000ms)...
```json
module: 'MMM-GPIO-Notifications',
config: {
'4': {
gpio_state: 1,
gpio_debounce: 0,
delay: 5000,
notifications: [
{
notification: 'SCREEN_ON',
payload: { 'forced': false }
},
module: 'MMM-Screen-Powersave-Notification',
config: {
delay: 30,
}
```

Tom-Hirschberger, 2022-10-17:
Hi,

it looks like you configured the wrong delay. The delay of the Gpio module causes triggers of the sensor to be ignored for a specific amount of time. The delay of the Screensave module configures how long the screen stays active if no more activity was detected.
So you simply need to set the delay of GPIO to a lower value (say 5000ms)...
```json
module: 'MMM-GPIO-Notifications',
config: {
'4': {
gpio_state: 1,
gpio_debounce: 0,
delay: 5000,
notifications: [
{
notification: 'SCREEN_ON',
payload: { 'forced': false }
},
module: 'MMM-Screen-Powersave-Notification',
config: {
delay: 30,
}
```

I will precise the howto at this point in the next days