Tom-Hirschberger/MMM-GPIO-Notifications #9: 0.7 stuck at postinstall

gonzonia, 2023-01-07:
After the recent MM upgrade I had to reinstall MMM-GPIO-Notifications. With the update to .7 I did a "git pull" followed by "npm install" but it seems to get stuck at > ./postinstall

gonzonia, 2023-01-07:
Looks like it eventually finished with an error. 
Here's the log. 
[2023-01-07T17_16_13_751Z-debug-0.log](https://github.com/Tom-Hirschberger/MMM-GPIO-Notifications/files/10366840/2023-01-07T17_16_13_751Z-debug-0.log)

gonzonia, 2023-01-07:
Tried a fresh install of MMM-GPIO-Notifications and got the following

> MMM-GPIO-Notifications@0.0.7 postinstall
> ./postinstall

./postinstall: line 8: ../../node_modules/.bin/electron-rebuild: Permission denied
npm ERR! code 126
npm ERR! path /home/pi/MagicMirror/modules/MMM-GPIO-Notifications
npm ERR! command failed
npm ERR! command sh -c ./postinstall

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2023-01-07T17_39_38_513Z-debug-0.log
[2023-01-07T17_39_38_513Z-debug-0.log](https://github.com/Tom-Hirschberger/MMM-GPIO-Notifications/files/10366849/2023-01-07T17_39_38_513Z-debug-0.log)

Tom-Hirschberger, 2023-01-07:
Hi and interesting...
Can you give me some more information about your environment, please?

Which architecture and operating system are you using?
Are you running a local version of MagicMirror or do you use it in a docker container?

Can you please change to the root directory of MagicMirror (usually /home/pi/MagicMirror or something similar), run the following command and paste the output here?
```bash
npm install @electron/rebuild
```

Can you please change to the directory "MagicMirror/node_modules/.bin", list the directory content and paste the output here?

gonzonia, 2023-01-07:
Raspberry Pi 4 Model B Rev 1.1
Raspbian GNU/Linux 10 (buster)
Local version

I've tried both 'npm install' with both sudo and without. 

npm install @electron/rebuild worked 

```
up to date, audited 819 packages in 6s

106 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

```

File List attached

[filelist.txt](https://github.com/Tom-Hirschberger/MMM-GPIO-Notifications/files/10367261/filelist.txt)

Tom-Hirschberger, 2023-01-07:
You are using the "normal" pi user, right?

So you may try the following steps...

First make. Sure that running with `sudo ` did not mess up any permissions:

```
cd /home/pi/MagicMirror
chown -R pi:users node_modules modules/MMM-GPIO-Notifications
```

As electron-rebuild seems to be installed correctly now you can run the postinstall script again:

```
cd /home/pi/MagicMirror/modules/MMM-GPIO-Notifications
./postinstall
```

gonzonia, 2023-01-08:
Yes. Normal pi user. 

Still get this 
./postinstall: line 8: ../../node_modules/.bin/electron-rebuild: Permission denied

Tom-Hirschberger, 2023-01-08:
As your file listing tells that the file exists next step is to check the permissions. Can you run
```
cd /home/pi/MagicMirror/node_modules/.bin
ls -l
```

To list the file again, please. This time the files are listed as table with the owner and permissions shown

gonzonia, 2023-01-08:
[binlist.txt](https://github.com/Tom-Hirschberger/MMM-GPIO-Notifications/files/10368663/binlist.txt)

Tom-Hirschberger, 2023-01-08:
Sorry for the iterative process but now I need a listing of the electron-rebuild directory:

```
cd /home/pi/MagicMirror/node_modules/@electron/rebuild/lib/src/
ls -l
```

gonzonia, 2023-01-08:
[srcfiles.txt](https://github.com/Tom-Hirschberger/MMM-GPIO-Notifications/files/10368735/srcfiles.txt)

gonzonia, 2023-01-08:
No need to apologize. I appreciated the time you're putting into this!

Tom-Hirschberger, 2023-01-08:
Perfect. It looks like permission issue. As the files are owned by the root user you are not allowed to run the scripts as user pi.

```
cd /home/pi/MagicMirror/node_modules
sudo chown -R pi:users *
```

And running postinstall as not root user should do the trick

gonzonia, 2023-01-08:
Still the same problem. I think you're right though, it's a permission issue somewhere. Just need to figure out where.

Tom-Hirschberger, 2023-01-08:
I think the following happend...
First there was a problem with installing @electron-rebuild with npm in the postinstall script. As you run it again with sudo the permissions got messed up.

I did not want to use the hammer but i think the only way to restore permissions will be to set ownership of all files in the MagicMirror directory back to user pi:

```
chown -R pi:users /home/pi/MagicMirror
```

And then try to run postinstall again or the electron-rebuild command directly:
```
cd /home/pi/MagicMirror/modules/MMM-GPIO-Notifications
../../node_modules/.bin/electron-rebuild
```

gonzonia, 2023-01-08:
I hate to say it....

pi@MagicMirror:~ $ sudo chown -R pi:users /home/pi/MagicMirror
pi@MagicMirror:~ $ cd /home/pi/MagicMirror/modules/MMM-GPIO-Notifications
pi@MagicMirror:~/MagicMirror/modules/MMM-GPIO-Notifications $ ../../node_modules/.bin/electron-rebuild
-bash: ../../node_modules/.bin/electron-rebuild: Permission denied

Same with ./postinstall

Tom-Hirschberger, 2023-01-08:
I found a difference in your file listing and mine one in the permissions. In my case the "cli.js" file of electron-rebuild has execution permissions. In your setup not.
This command will fix this:
```
chmod a+x /home/pi/MagicMirror/node_modules/@electron/rebuild/lib/src/cli.js
```

gonzonia, 2023-01-08:
That did it!! Thanks again for the time to troubleshoot!

closed 2023-01-08 by gonzonia

Tom-Hirschberger, 2023-01-08:
Your welcome. I do not know what caused the inital problems with installing electron-rebuild by npm but make sure to not run the `npm install` with sudo anymore ;-)

gonzonia, 2023-01-08:
Thanks. I probably got a "permission denied" somewhere else and used that to fix it. I know and understand enough just to cause myself problems.