Tom-Hirschberger/MMM-GPIO-Notifications #14: [Feature Request] Rotary Encoder Support

spxhub, 2023-06-02:
Hey, would it be possible to add support for rotary encoders?
Like currently always two notifications are sent as both pins are getting pulled high and low per single click.
It would be great to only send notification A when rotating CW and only notification B when rotation CCW.
I think having like a global gpio_debounce value where if ANY pin changes the state during this period after the last event the new event will be ignored could kinda work as depending on the direction of rotation one pin gets pulled before the other.

Tom-Hirschberger, 2023-06-03:
Hi,

i do not have any experience with rotary encoders but i do not think it is a big deal to add support.
Which model do you use?

spxhub, 2023-06-03:
I use a KY-040 model.
It seems like in the https://github.com/Ax-LED/MMM-Navigate module it is already implemented. Maybe you can lend from there?

Or maybe this helps:
https://www.kernel.org/doc/html/latest/input/devices/rotary-encoder.html
https://thepihut.com/blogs/raspberry-pi-tutorials/how-to-use-a-rotary-encoder-with-the-raspberry-pi

Tom-Hirschberger, 2023-06-03:
I ordered a view KY-040 a couple of seconds ago but it will take me some time to implement it.

What exactly do you want to do?
Send different notifications for ClockWise, CounterClockWise and Push?
Do you need to have the number of steps the rotary has been turned or do you want to send the notifications for each step?

spxhub, 2023-06-03:
Oh, cool! Thanks!
I only need notifications for CW and CCW. Push is already working. The idea is to send notifications to MMM-Carousel to navigate through different slides. But of course there are more use cases with other modules as well.

Tom-Hirschberger, 2023-06-17:
My sensors received a couple of days ago and today i had time for first investigations.
I will need to refactor some data structures and functions in the module to support rotaries. 
The main difference is that i do not only react to single state changes as i do currently but keep track of the last states of the two GPIOs used by each rotary.

I am planning to change the config only slightly to something like:

```json5
    {
        module: 'MMM-GPIO-Notifications',
        config: {
            '5,6': {
              notifications_cw: [
                {
                  'notification': 'ROTARY_TURNED_CLOCK_WISE',
                  //...
                }
              ], 
              notifications_ccw: [
                {
                  'notification': 'ROTARY_TURNED_COUNTER_CLOCK_WISE',
                  //...
                }
              ]
            }
        }
    },
```

Tom-Hirschberger, 2023-06-18:
Just released version 0.1.0 of the module.
It is possible to use rotaries like in the config example above now.
The readme contains a section of how to configure the rotary in more detail.

closed 2023-06-18 by Tom-Hirschberger

spxhub, 2023-06-18:
Cool, thank you! I've just updated and it's working great!

BTW: In your examples in the README you sometimes put _notification_ , _payload_ or _profile_ in ' ' (single quotation marks) and sometimes you don't. For the sake of better readability I would go without.

And your headline of the rotary encoder section is "Ratary encoder" (typo) ;-)

Tom-Hirschberger, 2023-06-18:
Great to see that it works.
I corrected the spelling mistake (and some more) and made the syntax uniform in the last commit.