Tom-Hirschberger/MMM-GPIO-Notifications #15: Is python required

mholzma, 2023-06-08:
I am trying to run 'npm install' and getting errors that I need to provide python. My Docker Image for MagicMirror doesn't include python by default it seems.

Any suggests/recommendations? I can include the error log if it would help.

mholzma, 2023-06-09:
Nevermind, I figured that part out by installing a version that has python.
However I still get the message "Skipping Pin registration because GPIO is not acessible!"

I've tried loading onoff to see if that fixed it but it doesn't seem to.

mholzma, 2023-06-09:
I believe I have sort of solved the issue. Other than having to get onoff installed via npm, I also had to set a variable in my docker container to allow access to the GPIO pins:
-v /sys:/sys 
Interestingly, the MMM-PIR-Sensor-Lite module does not require this as it is using python and RPi.GPIO to access the pins.

Tom-Hirschberger, 2023-06-10:
Hi and sorry for my delayed answer. I am on vacation at the moment and my Internet connection is not the best.

The module itself does not require Python but I need to use electron-rebuild to recompile the library for Gpio access. As electron-rebuild requires Python it is a dependency during npm install but not during runtime.

Great that you figured out a solution by yourself.
I use the "fat" image of karsten13 for my development. This contains Python.

My suggestion of how to fix the GPIO access would have been to run the container in privileged mode but I think your solution is the better and safer one.
Maybe I will add a hint about the problems/solutions about the use of docker with this module to the Readme.

Tom-Hirschberger, 2023-06-17:
I added some information of how to get the module running from within containers. Currently the changes are in the development branch but i will merge them with the next release.

Thank you for your investigation of the problem.
I will close the ticket then.

closed 2023-06-17 by Tom-Hirschberger