Tom-Hirschberger/MMM-GPIO-Notifications #18: Issues switching HDMI off on a PI3b

Matzeonom, 2024-02-12:
Hi Tom,

thank you for the description. 
But just wanted to share my problem using all with my Pi3b, where the command
**vcgencmd display_power 0**
is obviously not working.

After some "googling" I found this explaination:
https://forums.raspberrypi.com/viewtopic.php?t=331817

Tried out 
**xrandr --output HDMI-1 --off**
which works fine, but is not content of your MMM-GPIO-Notifications.

Now trying my best via this description:
https://github.com/paviro/MMM-PIR-Sensor

Hope, that helps others...

Regards
Matzeonom

Tom-Hirschberger, 2024-02-12:
Hi,

you are right vcgencmd is not working with the default graphics stack of Bullseye and Bookworm.
I wrote a wrapper which can be used with my Screensave module...

(Wrapper)[https://github.com/Tom-Hirschberger/MMM-Screen-Powersave-Notification/tree/master/controlScripts]

Tom-Hirschberger, 2024-02-12:
Hi,

you are right vcgencmd is not working with the default graphics stack of Bullseye and Bookworm.
I wrote a wrapper which can be used with my Screensave module...

(Wrapper)[https://github.com/Tom-Hirschberger/MMM-Screen-Powersave-Notification/tree/master/controlScripts]

The Readme of the Screensave module shows how to use the wrapper

Matzeonom, 2024-03-10:
Hi Tom,

finally, I managed to follow the above description - IT WORKED !

Thank you very much and kind support - viele Grüße
Matzeonom

closed 2024-03-10 by Matzeonom