Tom-Hirschberger/MMM-GPIO-Notifications #20: GPIO out and Pi5 support

Aergernis, 2024-03-14:
Hi,

is it possible to control GPIO Pins to switch relais, or do you have a Module for that?
And will it support Pi5?

Aergernis, 2024-03-17:
I've tried wiht a Pi5 and it can't register pins. Used just default config for MM and this module.
I also tested it with a Pi4 and latest bookworm install and it also can't register the pins.
```
0|MagicMirror  | [17.03.2024 13:34.27.986] [LOG]   MMM-GPIO-Notifications: Registering pin: 4
0|MagicMirror  | [17.03.2024 13:34.27.992] [ERROR] Whoops! There was an uncaught exception...
0|MagicMirror  | [17.03.2024 13:34.27.999] [ERROR] Error: EINVAL: invalid argument, write
0|MagicMirror  |     at Object.writeSync (node:fs:936:3)
0|MagicMirror  |     at Object.writeFileSync (node:fs:2330:26)
0|MagicMirror  |     at exportGpio (/home/philipp/MagicMirror/modules/MMM-GPIO-Notifications/node_modules/onoff/onoff.js:18:8)
0|MagicMirror  |     at new Gpio (/home/philipp/MagicMirror/modules/MMM-GPIO-Notifications/node_modules/onoff/onoff.js:172:36)
0|MagicMirror  |     at Class.registerSinglePin (/home/philipp/MagicMirror/modules/MMM-GPIO-Notifications/node_helper.js:296:30)
0|MagicMirror  |     at Class.socketNotificationReceived (/home/philipp/MagicMirror/modules/MMM-GPIO-Notifications/node_helper.js:457:10)
0|MagicMirror  |     at Socket.<anonymous> (/home/philipp/MagicMirror/js/node_helper.js:104:11)
0|MagicMirror  |     at Socket.emit (node:events:514:28)
0|MagicMirror  |     at Socket.emitUntyped (/home/philipp/MagicMirror/node_modules/socket.io/dist/typed-events.js:69:22)
0|MagicMirror  |     at /home/philipp/MagicMirror/node_modules/socket.io/dist/socket.js:704:39
0|MagicMirror  |     at process.processTicksAndRejections (node:internal/process/task_queues:77:11) {
0|MagicMirror  |   errno: -22,
0|MagicMirror  |   syscall: 'write',
0|MagicMirror  |   code: 'EINVAL'
0|MagicMirror  | }
```
Also MMM-Notification-To-Command is not listed in MMPM, would be nice to find it there without adding it manualy

Tom-Hirschberger, 2024-03-17:
I read about a different numbering of the pins but the library I am using still should be functional on the Pi5.

Are you sure your user is part of the GPIO group?

You can check it with the `groups` command.

I think i never added the module to the wiki that's why it is missing in MMPM.

Aergernis, 2024-03-17:
```
id
uid=1000(philipp) gid=1000(philipp) groups=1000(philipp),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),993(gpio),994(i2c),995(spi)
```
```
groups
philipp adm dialout cdrom sudo audio video plugdev games users input render netdev lpadmin gpio i2c spi
```

Tom-Hirschberger, 2024-03-17:
The groups look good to me. 
Maybe this link https://forums.raspberrypi.com/viewtopic.php?t=359302 is helpful.
Looks like you need to use a total different numbering.

Aergernis, 2024-03-17:
seems to be a old problem and was fixed in newer kernel. pinout seems to be the same for Pi4 and co....

```
Description        : Raspberry Pi 5B rev 1.0
Revision           : d04170
SoC                : BCM2712
RAM                : 8GB
Storage            : MicroSD
USB ports          : 4 (of which 2 USB3)
Ethernet ports     : 1 (1000Mbps max. speed)
Wi-fi              : True
Bluetooth          : True
Camera ports (CSI) : 2
Display ports (DSI): 2

,--------------------------------.
| oooooooooooooooooooo J8   : +====
| 1ooooooooooooooooooo      : |USB2
|  Wi  Pi Model 5B  V1.0  fan +====
|  Fi     +---+      +---+       |
|         |RAM|      |RP1|    +====
||p       +---+      +---+    |USB3
||c      -------              +====
||i        SoC      |c|c J14     |
(        -------  J7|s|s 12 +======
|  J2 bat   uart   1|i|i oo |   Net
| pwr\..|hd|...|hd|o|1|0    +======
`-| |-1o|m0|---|m1|--------------'

J8:
   3V3  (1) (2)  5V
 GPIO2  (3) (4)  5V
 GPIO3  (5) (6)  GND
 GPIO4  (7) (8)  GPIO14
   GND  (9) (10) GPIO15
GPIO17 (11) (12) GPIO18
GPIO27 (13) (14) GND
GPIO22 (15) (16) GPIO23
   3V3 (17) (18) GPIO24
GPIO10 (19) (20) GND
 GPIO9 (21) (22) GPIO25
GPIO11 (23) (24) GPIO8
   GND (25) (26) GPIO7
 GPIO0 (27) (28) GPIO1
 GPIO5 (29) (30) GND
 GPIO6 (31) (32) GPIO12
GPIO13 (33) (34) GND
GPIO19 (35) (36) GPIO16
GPIO26 (37) (38) GPIO20
   GND (39) (40) GPIO21

J2:
RUN (1)
GND (2)

J7:
COMPOSITE (1)
      GND (2)

J14:
TR01 TAP (1) (2) TR00 TAP
TR03 TAP (3) (4) TR02 TAP

For further information, please refer to https://pinout.xyz/

```

Tom-Hirschberger, 2024-03-17:
If I do understand correctly you will need to call

`grep -s . /sys/class/gpio/gpiochip*/*`

And check for the base number which is 512 in the post. 

Some Kernels add the GPIO number to the base some substract it of the base. 

So if your base is 512 the GPIO 4 will either be 508 or 512.

But I will have to look carefully it the onoff library will be updated or if I will have to switch to a different one in the future. 

Main problem is that there is no full replacement at the moment.

Aergernis, 2024-03-17:
I've tested a little..... gpio 18 and 27 are working, but i can not get 4 to work. I'll try again next week, maybe i'll find something

Tom-Hirschberger, 2024-03-20:
Looks like there is a problem with kernel 6.6 that has been released last month.
A lot of libraries are broken a the moment (Python and Javascript) some of them will never be fixed.

I followed this guide https://www.markjunghanns.de/de_DE/index.php/2017/12/27/raspberry-pi-kernel-upgraden-downgraden/ and downgraded my kernel to the previous 6.1 version with

```
sudo rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f
```

After this everything is working on my Bookworm system.
My fully updated  Bullseye system still uses 6.1 kernel so no problem there.

Aergernis, 2024-03-21:
Thanks for your help so fare.
Tried with Kernel 6.1, but GPIO4 (Pin7) is set to Pull high when the module starts, but i need it to be Pull down.

Tom-Hirschberger, 2024-03-21:
> [Verfolgungsbild][https://github.com/notifications/beacon/ALMFULAGPADXIQFJOYDVMSDYZL5YTA5CNFSM6AAAAABEWJA3BGWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTX65WZO.gif]

Hi,

you can set the pull up and pull down in the /boot/config.txt but I do not know if it is still working

Aergernis, 2024-03-22:
Tried to install your new release, but it's not installaing. `node_modules` folder is not created.

```
0 verbose cli /usr/local/bin/node /usr/local/bin/npm
1 info using npm@10.1.0
2 info using node@v20.8.0
3 timing npm:load:whichnode Completed in 2ms
4 timing config:load:defaults Completed in 2ms
5 timing config:load:file:/usr/local/lib/node_modules/npm/npmrc Completed in 0ms
6 timing config:load:builtin Completed in 1ms
7 timing config:load:cli Completed in 2ms
8 timing config:load:env Completed in 0ms
9 timing config:load:file:/home/philipp/MagicMirror/modules/MMM-GPIO-Notifications/.npmrc Completed in 0ms
10 timing config:load:project Completed in 3ms
11 timing config:load:file:/home/philipp/.npmrc Completed in 0ms
12 timing config:load:user Completed in 0ms
13 timing config:load:file:/usr/local/etc/npmrc Completed in 0ms
14 timing config:load:global Completed in 0ms
15 timing config:load:setEnvs Completed in 0ms
16 timing config:load Completed in 9ms
17 timing npm:load:configload Completed in 10ms
18 timing config:load:flatten Completed in 2ms
19 timing npm:load:mkdirpcache Completed in 1ms
20 timing npm:load:mkdirplogs Completed in 0ms
21 verbose title npm install
22 verbose argv "install"
23 timing npm:load:setTitle Completed in 1ms
24 timing npm:load:display Completed in 1ms
25 verbose logfile logs-max:10 dir:/home/philipp/.npm/_logs/2024-03-22T19_07_35_911Z-
26 verbose logfile /home/philipp/.npm/_logs/2024-03-22T19_07_35_911Z-debug-0.log
27 timing npm:load:logFile Completed in 9ms
28 timing npm:load:timers Completed in 0ms
29 timing npm:load:configScope Completed in 0ms
30 timing npm:load Completed in 38ms
31 timing arborist:ctor Completed in 1ms
32 timing arborist:ctor Completed in 0ms
33 silly logfile start cleaning logs, removing 1 files
34 silly logfile done cleaning log files
35 timing idealTree:init Completed in 15ms
36 timing idealTree:userRequests Completed in 0ms
37 silly idealTree buildDeps
38 silly fetch manifest opengpio@latest
39 http fetch GET 200 https://registry.npmjs.org/opengpio 1292ms (cache miss)
40 silly placeDep ROOT opengpio@1.0.11 OK for: MMM-GPIO-Notifications@0.2.0 want: latest
41 silly fetch manifest bindings@^1.5.0
42 silly fetch manifest debug@^4.3.4
43 silly fetch manifest node-addon-api@^7.0.0
44 http fetch GET 200 https://registry.npmjs.org/node-addon-api 72ms (cache miss)
45 http fetch GET 200 https://registry.npmjs.org/debug 155ms (cache revalidated)
46 http fetch GET 200 https://registry.npmjs.org/bindings 164ms (cache revalidated)
47 timing idealTree:#root Completed in 1469ms
48 silly placeDep ROOT bindings@1.5.0 OK for: opengpio@1.0.11 want: ^1.5.0
49 silly fetch manifest file-uri-to-path@1.0.0
50 silly placeDep ROOT debug@4.3.4 OK for: opengpio@1.0.11 want: ^4.3.4
51 silly fetch manifest ms@2.1.2
52 silly placeDep ROOT node-addon-api@7.1.0 OK for: opengpio@1.0.11 want: ^7.0.0
53 http fetch GET 200 https://registry.npmjs.org/file-uri-to-path 64ms (cache revalidated)
54 http fetch GET 200 https://registry.npmjs.org/ms 90ms (cache revalidated)
55 timing idealTree:node_modules/opengpio Completed in 98ms
56 silly placeDep ROOT file-uri-to-path@1.0.0 OK for: bindings@1.5.0 want: 1.0.0
57 timing idealTree:node_modules/bindings Completed in 2ms
58 silly placeDep ROOT ms@2.1.2 OK for: debug@4.3.4 want: 2.1.2
59 timing idealTree:node_modules/debug Completed in 3ms
60 timing idealTree:node_modules/file-uri-to-path Completed in 0ms
61 timing idealTree:node_modules/ms Completed in 0ms
62 timing idealTree:node_modules/node-addon-api Completed in 0ms
63 timing idealTree:buildDeps Completed in 1574ms
64 timing idealTree:fixDepFlags Completed in 0ms
65 timing idealTree Completed in 1590ms
66 timing reify:loadTrees Completed in 1591ms
[... 89 lines cut ...]
107 error compilation terminated.
107 error make: *** [opengpio.target.mk:111: Release/obj.target/opengpio/cpp/opengpio.o] Error 1
107 error gyp ERR! build error 
107 error gyp ERR! stack Error: `make` failed with exit code: 2
107 error gyp ERR! stack     at ChildProcess.onExit (/home/philipp/MagicMirror/node_modules/node-gyp/lib/build.js:203:23)
107 error gyp ERR! stack     at ChildProcess.emit (node:events:514:28)
107 error gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:294:12)
107 error gyp ERR! System Linux 6.6.22-v8-16k+
107 error gyp ERR! command "/usr/local/bin/node" "/home/philipp/MagicMirror/node_modules/.bin/node-gyp" "rebuild"
107 error gyp ERR! cwd /home/philipp/MagicMirror/modules/MMM-GPIO-Notifications/node_modules/opengpio
107 error gyp ERR! node -v v20.8.0
107 error gyp ERR! node-gyp -v v9.4.1
107 error gyp ERR! not ok
108 verbose exit 1
109 timing npm Completed in 3872ms
110 verbose unfinished npm timer reify 1711134456287
111 verbose unfinished npm timer reify:build 1711134458033
112 verbose unfinished npm timer build 1711134458033
113 verbose unfinished npm timer build:deps 1711134458034
114 verbose unfinished npm timer build:run:install 1711134458035
115 verbose unfinished npm timer build:run:install:node_modules/opengpio 1711134458035
116 verbose code 1
117 error A complete log of this run can be found in: /home/philipp/.npm/_logs/2024-03-22T19_07_35_911Z-debug-0.log

```

Tom-Hirschberger, 2024-03-22:
Looks like the libdpiod-dev package had not been installed.

Can you please got to the module directory, run the preinstall script manually and provide the output.

Additionally the output of
```
sudo apt list --installed ¦ grep gpio
```
would be great.

Aergernis, 2024-03-22:
`./preinstall`
```
Check for package libgpiod-dev and install if needed...missing
Installing missing package libgpiod-dev...
OK:1 http://archive.raspberrypi.com/debian bookworm InRelease
OK:2 http://deb.debian.org/debian bookworm InRelease
OK:3 http://deb.debian.org/debian-security bookworm-security InRelease
OK:4 http://deb.debian.org/debian bookworm-updates InRelease
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Alle Pakete sind aktuell.
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Vorgeschlagene Pakete:
  libgpiod-doc
Die folgenden NEUEN Pakete werden installiert:
  libgpiod-dev
0 aktualisiert, 1 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 58,4 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 352 kB Plattenplatz zusätzlich benutzt.
Holen:1 http://deb.debian.org/debian bookworm/main arm64 libgpiod-dev arm64 1.6.3-1+b3 [58,4 kB]
Es wurden 58,4 kB in 0 s geholt (632 kB/s).
Vormals nicht ausgewähltes Paket libgpiod-dev:arm64 wird gewählt.
(Lese Datenbank ... 252307 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../libgpiod-dev_1.6.3-1+b3_arm64.deb ...
Entpacken von libgpiod-dev:arm64 (1.6.3-1+b3) ...
libgpiod-dev:arm64 (1.6.3-1+b3) wird eingerichtet ...
done
Check for package gpiod and install if needed...Skipping. Already installed.
Calling converter script to gather info about the gpio chips, lines and names...
Will write results to: gpioinfo.json
Gathering infos with gpioinfo
Processing the output of gpioinfo
Writing result file
Got the info we need and got it converted.

```
`sudo apt list --installed | grep gpio`
```
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

gpiod/stable,now 1.6.3-1+b3 arm64  [installiert]
libgpiod-dev/stable,now 1.6.3-1+b3 arm64  [installiert]
libgpiod2/stable,now 1.6.3-1+b3 arm64  [Installiert,automatisch]
liblgpio1/stable,now 0.2.2-1~rpt1 arm64  [Installiert,automatisch]
libpigpio-dev/stable,now 1.79-1+rpt1 arm64  [Installiert,automatisch]
libpigpio1/stable,now 1.79-1+rpt1 arm64  [Installiert,automatisch]
libpigpiod-if-dev/stable,now 1.79-1+rpt1 arm64  [Installiert,automatisch]
libpigpiod-if1/stable,now 1.79-1+rpt1 arm64  [Installiert,automatisch]
libpigpiod-if2-1/stable,now 1.79-1+rpt1 arm64  [Installiert,automatisch]
pigpio-tools/stable,now 1.79-1+rpt1 arm64  [Installiert,automatisch]
pigpio/stable,now 1.79-1+rpt1 arm64  [installiert]
pigpiod/stable,now 1.79-1+rpt1 arm64  [Installiert,automatisch]
python3-gpiozero/stable,stable,now 2.0-1 all  [installiert]
python3-lgpio/stable,now 0.2.2-1~rpt1 arm64  [Installiert,automatisch]
python3-libgpiod/stable,now 1.6.3-1+b3 arm64  [installiert]
python3-pigpio/stable,stable,now 1.79-1+rpt1 all  [installiert]
python3-rpi.gpio/stable,now 0.7.1~a4-1+b4 arm64  [installiert]
raspi-gpio/stable,now 0.20231127 arm64  [installiert]
rpi.gpio-common/stable,now 0.7.1~a4-1+b4 arm64  [Installiert,automatisch]

```

Tom-Hirschberger, 2024-03-22:
I am pretty sure if you run npm install it will work now.
Do not know why the preinstall script is not getting called.

Added a hint to the readme a few seconds ago.

Aergernis, 2024-03-22:
i think `./preinstall` is called to late in install process. It is called after trying to create gpioinfo.json

```
Schwerwiegend: Zielpfad 'MMM-GPIO-Notifications' existiert bereits und ist kein leeres Verzeichnis.
Check for package libgpiod-dev and install if needed...Skipping. Already installed.
Check for package gpiod and install if needed...Skipping. Already installed.
Calling converter script to gather info about the gpio chips, lines and names...
Will write results to: gpioinfo.json
Gathering infos with gpioinfo
Processing the output of gpioinfo
Writing result file
Got the info we need and got it converted.

> MMM-GPIO-Notifications@0.2.0 preinstall
> ./preinstall

Check for package libgpiod-dev and install if needed...Skipping. Already installed.
Check for package gpiod and install if needed...Skipping. Already installed.
Calling converter script to gather info about the gpio chips, lines and names...
Will write results to: gpioinfo.json
Gathering infos with gpioinfo
Processing the output of gpioinfo
Writing result file
Got the info we need and got it converted.

added 6 packages, and audited 7 packages in 8s

found 0 vulnerabilities

```

Aergernis, 2024-03-22:
Maybe this will help https://github.com/npm/cli/issues/2660

Tom-Hirschberger, 2024-03-22:
As the converter script which creates the gpioinfo.json I do not think it is called to late.

It looks more like it does not get called the first time npm install is called and libgpiod-dev is missing.

Tom-Hirschberger, 2024-03-22:
> Maybe this will help https://github.com/npm/cli/issues/2660

You are right. Looks like exactly the same problem.

Aergernis, 2024-03-22:
it's installoing fine, but to get pin 4 as input pulled down i still had to set it in config.txt

gpioinfo without config.txt

```
gpiochip4 - 54 lines:
        line   0:     "ID_SDA"       unused   input  active-high
        line   1:     "ID_SCL"       unused   input  active-high
        line   2:      "GPIO2"       unused   input  active-high
        line   3:      "GPIO3"       unused   input  active-high
        line   4:      "GPIO4" "opengpio_4_4_watch" input active-high [used]
        line   5:      "GPIO5"       unused   input  active-high
        line   6:      "GPIO6"       unused   input  active-high
        line   7:      "GPIO7"       unused   input  active-high

```
with config.txt

```
gpiochip4 - 54 lines:
        line   0:     "ID_SDA"       unused   input  active-high
        line   1:     "ID_SCL"       unused   input  active-high
        line   2:      "GPIO2"       unused   input  active-high
        line   3:      "GPIO3"       unused   input  active-high
        line   4:      "GPIO4" "opengpio_4_4_watch" input active-high [used]
        line   5:      "GPIO5"       unused   input  active-high
        line   6:      "GPIO6"       unused   input  active-high
        line   7:      "GPIO7"       unused   input  active-high

```

it's both showing the same for the pin, but it's not wirking without the edit

Tom-Hirschberger, 2024-03-29:
I opened a feature request at the libraries repo to support pull-up and pull-down but it may take a moment to be implemented.

Can you do me a favour and post a complete output of your gpioinfo so I can add Raspberry 5 as a devide to the opengpio library?

If it is OK for you I will close the issue as long as the library does not support pull the pins to a state.

Aergernis, 2024-03-30:
```
gpiochip0 - 32 lines:
        line   0:          "-"       unused   input  active-high
        line   1: "2712_BOOT_CS_N" "spi10 CS0" output active-low [used]
        line   2: "2712_BOOT_MISO" unused input active-high
        line   3: "2712_BOOT_MOSI" unused input active-high
        line   4: "2712_BOOT_SCLK" unused input active-high
        line   5:          "-"       unused   input  active-high
        line   6:          "-"       unused   input  active-high
        line   7:          "-"       unused   input  active-high
        line   8:          "-"       unused   input  active-high
        line   9:          "-"       unused   input  active-high
        line  10:          "-"       unused   input  active-high
        line  11:          "-"       unused   input  active-high
        line  12:          "-"       unused   input  active-high
        line  13:          "-"       unused   input  active-high
        line  14:   "PCIE_SDA"       unused   input  active-high
        line  15:   "PCIE_SCL"       unused   input  active-high
        line  16:          "-"       unused   input  active-high
        line  17:          "-"       unused   input  active-high
        line  18:          "-"       unused   input  active-high
        line  19:          "-"       unused   input  active-high
        line  20:   "PWR_GPIO" "pwr_button"   input   active-low [used]
        line  21: "2712_G21_FS" unused input active-high
        line  22:          "-"       unused   input  active-high
        line  23:          "-"       unused   input  active-high
        line  24:     "BT_RTS"       unused   input  active-high
        line  25:     "BT_CTS"       unused   input  active-high
        line  26:     "BT_TXD"       unused   input  active-high
        line  27:     "BT_RXD"       unused   input  active-high
        line  28:      "WL_ON"  "wl_on_reg"  output  active-high [used]
        line  29:      "BT_ON"   "shutdown"  output  active-high [used]
        line  30: "WIFI_SDIO_CLK" unused input active-high
        line  31: "WIFI_SDIO_CMD" unused input active-high
gpiochip1 - 4 lines:
        line   0: "WIFI_SDIO_D0" unused input active-high
        line   1: "WIFI_SDIO_D1" unused input active-high
        line   2: "WIFI_SDIO_D2" unused input active-high
        line   3: "WIFI_SDIO_D3" unused input active-high
gpiochip2 - 17 lines:
        line   0:    "RP1_SDA"       unused   input  active-high
        line   1:    "RP1_SCL"       unused   input  active-high
        line   2:    "RP1_RUN" "RP1 RUN pin" output active-high [used]
        line   3: "SD_IOVDD_SEL" "vdd-sd-io" output active-high [used]
        line   4:  "SD_PWR_ON" "sd_vcc_reg"  output  active-high [used]
        line   5:  "SD_CDET_N"         "cd"   input   active-low [used]
        line   6:   "SD_FLG_N"       unused   input  active-high
        line   7:          "-"       unused   input  active-high
        line   8:  "2712_WAKE"       unused   input  active-high
        line   9: "2712_STAT_LED" "ACT" output active-low [used]
        line  10:          "-"       unused   input  active-high
        line  11:          "-"       unused   input  active-high
        line  12:   "PMIC_INT"       unused   input  active-high
        line  13: "UART_TX_FS"       unused   input  active-high
        line  14: "UART_RX_FS"       unused   input  active-high
        line  15:          "-"       unused   input  active-high
        line  16:          "-"       unused   input  active-high
gpiochip3 - 6 lines:
        line   0:  "HDMI0_SCL"       unused   input  active-high
        line   1:  "HDMI0_SDA"       unused   input  active-high
        line   2:  "HDMI1_SCL"       unused   input  active-high
        line   3:  "HDMI1_SDA"       unused   input  active-high
        line   4:   "PMIC_SCL"       unused   input  active-high
        line   5:   "PMIC_SDA"       unused   input  active-high
gpiochip4 - 54 lines:
        line   0:     "ID_SDA"       unused   input  active-high
        line   1:     "ID_SCL"       unused   input  active-high
        line   2:      "GPIO2"       unused   input  active-high
        line   3:      "GPIO3"       unused   input  active-high
        line   4:      "GPIO4" "opengpio_4_4_watch" input active-high [used]
[... 26 lines cut ...]
        line  31:   "HOST_SCL"       unused   input  active-high
        line  32:  "ETH_RST_N"  "phy-reset"  output   active-low [used]
        line  33:          "-"       unused   input  active-high
        line  34: "CD0_IO0_MICCLK" "cam0_reg" output active-high [used]
        line  35: "CD0_IO0_MICDAT0" unused input active-high
        line  36: "RP1_PCIE_CLKREQ_N" unused input active-high
        line  37:          "-"       unused   input  active-high
        line  38:    "CD0_SDA"       unused   input  active-high
        line  39:    "CD0_SCL"       unused   input  active-high
        line  40:    "CD1_SDA"       unused   input  active-high
        line  41:    "CD1_SCL"       unused   input  active-high
        line  42: "USB_VBUS_EN" unused output active-high
        line  43:   "USB_OC_N"       unused   input  active-high
        line  44: "RP1_STAT_LED" "PWR" output active-low [used]
        line  45:    "FAN_PWM"       unused  output  active-high
        line  46: "CD1_IO0_MICCLK" "cam1_reg" output active-high [used]
        line  47:  "2712_WAKE"       unused   input  active-high
        line  48: "CD1_IO1_MICDAT1" unused input active-high
        line  49: "EN_MAX_USB_CUR" unused output active-high
        line  50:          "-"       unused   input  active-high
        line  51:          "-"       unused   input  active-high
        line  52:          "-"       unused   input  active-high
        line  53:          "-"       unused   input  active-high

```

closed 2024-04-16 by Tom-Hirschberger