Tom-Hirschberger/MMM-GPIO-Notifications #21: Getting an errno: -22, 0|mm       |   syscall: 'write', 0|mm       |   code: 'EINVAL' error at MM start up and not detecting input

bayfacalendar, 2024-03-18:
`at exportGpio (/home/Admin/MagicMirror/modules/MMM-GPIO-Notifications/node_modules/onoff/onoff.js:18:8)
0|mm       |     at new Gpio (/home/Admin/MagicMirror/modules/MMM-GPIO-Notifications/node_modules/onoff/onoff.js:172:36)
0|mm       |     at Class.registerSinglePin (/home/Admin/MagicMirror/modules/MMM-GPIO-Notifications/node_helper.js:296:30)
0|mm       |     at Class.socketNotificationReceived (/home/Admin/MagicMirror/modules/MMM-GPIO-Notifications/node_helper.js:457:10)
0|mm       |     at Socket.<anonymous> (/home/Admin/MagicMirror/js/node_helper.js:104:11)
0|mm       |     at Socket.emit (node:events:514:28)
0|mm       |     at Socket.emitUntyped (/home/Admin/MagicMirror/node_modules/socket.io/dist/typed-events.js:69:22)
0|mm       |     at /home/Admin/MagicMirror/node_modules/socket.io/dist/socket.js:704:39
0|mm       |     at process.processTicksAndRejections (node:internal/process/task_queues:77:11) {
0|mm       |   errno: -22,
0|mm       |   syscall: 'write',
0|mm       |   code: 'EINVAL'
0|mm       | }`

Tom-Hirschberger, 2024-03-18:
> [Verfolgungsbild][https://github.com/notifications/beacon/ALMFULD5A4TRUF6KFTMOBWTYY4OUNA5CNFSM6AAAAABE4AHYPGWGG33NNVSW45C7OR4XAZNFJFZXG5LFVJRW63LNMVXHIX3JMTHIFM6E2I.gif]

Hi,

It looks like there is a general issue with the onoff library I am using and current installations of Bookworm.
There is a open similar issue.
I currently do not have solution and not much time to debug.
Maybe it is a kernel bug and will be fixed shortly.

bayfacalendar, 2024-03-18:
Thank you! was going crazy trying to figure this out!

Tom-Hirschberger, 2024-03-20:
Looks like there is a problem with kernel 6.6 that has been released last month.
A lot of libraries are broken a the moment (Python and Javascript) some of them will never be fixed.

I followed this guide https://www.markjunghanns.de/de_DE/index.php/2017/12/27/raspberry-pi-kernel-upgraden-downgraden/ and downgraded my kernel to the previous 6.1 version with

```
sudo rpi-update 5fc4f643d2e9c5aa972828705a902d184527ae3f
```

After this everything is working on my Bookworm system.
My fully updated  Bullseye system still uses 6.1 kernel so no problem there.

Tom-Hirschberger, 2024-03-22:
released version 0.2.0 of the module which uses a different library and should work with all raspberry versions without any hacks or downgrade of the kernel.

The new library uses the state of the art libgpiod bindings instead of the deprecated /sys/class/gpio sysfs. Should be fine for future releases, too.

Some tools and the libgpiod-dev package are required to be installed to get the module work. But if you do a `npm install` after the update they will be installed automatically.

Tom-Hirschberger, 2024-03-22:
released version 0.2.0 of the module which uses a different library and should work with all raspberry versions without any hacks or downgrade of the kernel.

The new library uses the state of the art libgpiod bindings instead of the deprecated /sys/class/gpio sysfs. Should be fine for future releases, too.

Some tools and the libgpiod-dev package are required to be installed to get the module work. But if you do a npm install after the update they will be installed automatically

bayfacalendar, 2024-03-22:
I'll do an update today and let you know!

Thanks

bayfacalendar, 2024-03-23:
Hey Tom I upgraded to version 0.2.0, but nothing happens.

in my pm2 logs i see 0|mm  | [23.03.2024 14:55.51.016] [LOG]   MMM-GPIO-Notifications: Registering pin: 14.

But its not working when I press the button, and gpioinfo shows the pin as unused.

Any Ideas?

Tom-Hirschberger, 2024-03-23:
Hi,

do you see any other pins used by opengpio in gpioinfo while MagicMirror is running?
Maybe my mapping file is garbage.

Do you see any errors in the error log file (not the output log file)?

bayfacalendar, 2024-03-23:
I see pin3 which i have setup as a power button.
 line   3:       "SCL1"   "shutdown"   input   active-low [used]

no other errors in the pm2 logs, or in the syslog / journal.

Tom-Hirschberger, 2024-03-23:
Can you post the content of your gpioinfo.json file please?

bayfacalendar, 2024-03-23:
{
    "gpios": {
        "ID_SDA": [
            0,
            0
        ],
        "ID_SCL": [
            0,
            1
        ],
        "SDA1": [
            0,
            2
        ],
        "SCL1": [
            0,
            3
        ],
        "GPIO_GCLK": [
            0,
            4
        ],
        "GPIO5": [
            0,
            5
        ],
        "GPIO6": [
            0,
            6
        ],
        "SPI_CE1_N": [
            0,
            7
        ],
        "SPI_CE0_N": [
            0,
            8
        ],
        "SPI_MISO": [
            0,
            9
        ],
        "SPI_MOSI": [
            0,
            10
        ],
        "SPI_SCLK": [
            0,
            11
        ],
        "GPIO12": [
            0,
            12
        ],
        "GPIO13": [
            0,
            13
        ],
        "TXD1": [
            0,
            14
        ],
        "RXD1": [
            0,
            15
        ],
        "GPIO16": [
            0,
            16
        ],
[... 245 lines cut ...]
            "RGMII_RXCLK": 46,
            "RGMII_RXCTL": 47,
            "RGMII_RXD0": 48,
            "RGMII_RXD1": 49,
            "RGMII_RXD2": 50,
            "RGMII_RXD3": 51,
            "RGMII_TXCLK": 52,
            "RGMII_TXCTL": 53,
            "RGMII_TXD0": 54,
            "RGMII_TXD1": 55,
            "RGMII_TXD2": 56,
            "RGMII_TXD3": 57
        },
        "gpiochip1": {
            "BT_ON": 0,
            "WL_ON": 1,
            "PWR_LED_OFF": 2,
            "GLOBAL_RESET": 3,
            "VDD_SD_IO_SEL": 4,
            "CAM_GPIO": 5,
            "SD_PWR_ON": 6,
            "SD_OC_N": 7
        }
    }
}

Tom-Hirschberger, 2024-03-23:
It looks like GPIO 14 is used by some other function as it is not part of the gpioinfo.json.

I will need to add a message for this case to the module but it will not solve your problem.

Can you switch to any GPIO which is named in file as GPIOX where X is the number of the GPIO?

bayfacalendar, 2024-03-23:
amazing! Switched it to GPIO16.

working now. Thank you so much!

Tom-Hirschberger, 2024-03-23:
Great.

I will change the code in the next days so any pin can be used even if the name is not starting with GPIO.

Tom-Hirschberger, 2024-03-29:
As of Version 0.2.1 you will be able to use GPIOs by their name which is shown in gpioinfo.
I will close the issue as the intial problem should be solved.
Feel free to open a new one or reopen it if you face any problems.

closed 2024-03-29 by Tom-Hirschberger