Tom-Hirschberger/MMM-GPIO-Notifications #25: GPIO state after stop - module-restart demands motion in front of the sensor ?

Matzeonom, 2024-04-01:
Dear Tom,
again thanks for the effort spent on this module - had nice Easter finalizing the project.

But - there seems to be the issue, that after stopping the module, that the restart runs into an error if there had been no motion in front of the sensor right befort restart. This is the protocoll:

[01.04.2024 11:07.42.978] [LOG]   Launching application.
[01.04.2024 11:07.55.159] [LOG]   MMM-GPIO-Notifications: Trying to registering pin: 17
[01.04.2024 11:07.55.161] [LOG]   MMM-GPIO-Notifications: Using chip and lane info of GPIO: GPIO17
[01.04.2024 11:07.55.163] [LOG]   MMM-GPIO-Notifications: Watched pin: 17 has low state delay of 10000!
[01.04.2024 11:07.55.164] [LOG]   MMM-GPIO-Notifications: Watched pin: 17 has high state delay of 10000!
terminate called after throwing an instance of 'std::system_error'
  what():  error requesting GPIO lines: Das Gerät oder die Ressource ist belegt
/home/user/MagicMirror/node_modules/electron/dist/electron exited with signal SIGABRT

I am just "user" and no pro - any proposal, how to solve (instead of dancing in front of the sensor, before every restart ;-))

Enjoy "Ostermontag", and thanks
matzeonom

Tom-Hirschberger, 2024-04-01:
> [Verfolgungsbild][https://github.com/notifications/beacon/ALMFULHRQFQL67MFPRCOW6DY3ESYPA5CNFSM6AAAAABFRK6S3GWGG33NNVSW45C7OR4XAZNFJFZXG5LFVJRW63LNMVXHIX3JMTHIIMVMX4.gif]

Hi,

and happy Easter.
It might be a problem that I did not free the resources fast enough during the shutdown of the module.
I checked the code and there might be a easy solution but I not know when I will have some spare time to implement it.
You can try to wait a few seconds between the stop and the start of MagicMirror as a workaround.

Matzeonom, 2024-04-05:
Hi Tom et al,

I don't think, that is the reason. This is, what the log-file says:

0|mm       | [1964:0405/195637.607417:ERROR:browser_main_loop.cc(278)] Gdk: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed
0|mm       | [1964:0405/195637.621132:ERROR:browser_main_loop.cc(278)] Gdk: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed
0|mm       | [1964:0405/195637.627125:ERROR:browser_main_loop.cc(278)] Gdk: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed
0|mm       | [2024-04-05 19:57:43.776] [ERROR] Error when stopping node_helper for module MMM-GPIO-Notifications: 
0|mm       | [2024-04-05 19:57:43.785] [ERROR] ReferenceError: self is not defined
0|mm       |     at Class.stop (/home/user/MagicMirror/modules/MMM-GPIO-Notifications/node_helper.js:67:25)
0|mm       |     at App.stop (/home/user/MagicMirror/js/app.js:307:35)
0|mm       |     at App.<anonymous> (/home/user/MagicMirror/js/electron.js:171:13)
0|mm       |     at App.emit (node:events:514:28)
0|mm       |     at appMethod (node:electron/js2c/browser_init:2:23592)
0|mm       |     at Object.execute (node:electron/js2c/browser_init:2:27684)
0|mm       |     at MenuItem.click (node:electron/js2c/browser_init:2:29513)
0|mm       |     at a._executeCommand (node:electron/js2c/browser_init:2:34987) 
0|mm       | /home/user/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
0|mm       | /home/user/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT

Somehow, the restart of the display goes wrong, eventhough the module is still working. 
Funny enough, it was working some days ago, no clue, why it suddenly stops. Sensor itself is ok.

Any help is highly appreciated.

A good weekend everybody - Cheers
matzeonom

Tom-Hirschberger, 2024-04-05:
Hi,

that's a new problem ;-)

Just did a quick fix a few seconds ago.

Can you please try the latest version.

Thanks

Matzeonom, 2024-04-06:
Good morning,

and thank you for the quick action. Unfortunately and by accident, I have killed my whole MMM installation.
Had to completely reinstall, but is still not working. Log is like this:

0|mm       | [2024-04-06 09:20:19.362] [ERROR] Error when stopping node_helper for module MMM-GPIO-Notifications: 
0|mm       | [2024-04-06 09:20:19.365] [ERROR] TypeError: self.registeredPins is not iterable
0|mm       |     at Class.stop (/home/user/MagicMirror/modules/MMM-GPIO-Notifications/node_helper.js:68:25)
0|mm       |     at App.stop (/home/user/MagicMirror/js/app.js:307:35)
0|mm       |     at App.<anonymous> (/home/user/MagicMirror/js/electron.js:171:13)
0|mm       |     at App.emit (node:events:514:28)
0|mm       |     at appMethod (node:electron/js2c/browser_init:2:23592)
0|mm       |     at Object.execute (node:electron/js2c/browser_init:2:27684)
0|mm       |     at MenuItem.click (node:electron/js2c/browser_init:2:29513)
0|mm       |     at a._executeCommand (node:electron/js2c/browser_init:2:34987) 

And now, even the Notification-Module is now showing any signals from the sensor...

For your information: I am using PM2 for automatic start and the PI is connected via WLAN, which is not in any case very stable.
Could that influence the performance ?

Enjoy the sunny weekend, regards
matzeonom

Tom-Hirschberger, 2024-04-06:
Hi and sorry,

not a WiFi problem definitely...Another bug programmed by me ;-)

Just released 0.2.4 which addresses this issue.

I do not know why but in my Mirror i do not see any console messages during shutdown thats way i missed this issue.

Tom-Hirschberger, 2024-04-06:
Released 0.2.5 a couple of seconds ago because of a other issue.

Can you please provide the output of:

```
cat /proc/device-tree/model
```

and
```
gpioinfo
```

to me.

Matzeonom, 2024-04-06:
Hi Tom,
good news first - it works again :-)

The desired information:
Raspberry Pi 3 Model B Plus Rev 1.4

        line   0:     "ID_SDA"       unused   input  active-high 
	line   1:     "ID_SCL"       unused   input  active-high 
	line   2:       "SDA1"       unused   input  active-high 
	line   3:       "SCL1"       unused   input  active-high 
	line   4:  "GPIO_GCLK"       unused   input  active-high 
	line   5:      "GPIO5"       unused   input  active-high 
	line   6:      "GPIO6"       unused   input  active-high 
	line   7:  "SPI_CE1_N"       unused   input  active-high 
	line   8:  "SPI_CE0_N"       unused   input  active-high 
	line   9:   "SPI_MISO"       unused   input  active-high 
	line  10:   "SPI_MOSI"       unused   input  active-high 
	line  11:   "SPI_SCLK"       unused   input  active-high 
	line  12:     "GPIO12"       unused   input  active-high 
	line  13:     "GPIO13"       unused   input  active-high 
	line  14:       "TXD1"       unused   input  active-high 
	line  15:       "RXD1"       unused   input  active-high 
	line  16:     "GPIO16"       unused   input  active-high 
	line  17:     "GPIO17"       unused   input  active-high 
	line  18:     "GPIO18"       unused   input  active-high 
	line  19:     "GPIO19"       unused   input  active-high 
	line  20:     "GPIO20"       unused   input  active-high 
	line  21:     "GPIO21"       unused   input  active-high 
	line  22:     "GPIO22"       unused   input  active-high 
	line  23:     "GPIO23"       unused   input  active-high 
	line  24:     "GPIO24"       unused   input  active-high 
	line  25:     "GPIO25"       unused   input  active-high 
	line  26:     "GPIO26"       unused   input  active-high 
	line  27:     "GPIO27"       unused   input  active-high 
	line  28: "HDMI_HPD_N"        "hpd"   input   active-low [used]
	line  29: "STATUS_LED_G" "ACT" output active-high [used]
	line  30:       "CTS0"       unused   input  active-high 
	line  31:       "RTS0"       unused   input  active-high 
	line  32:       "TXD0"       unused   input  active-high 
	line  33:       "RXD0"       unused   input  active-high 
	line  34:    "SD1_CLK"       unused   input  active-high 
	line  35:    "SD1_CMD"       unused   input  active-high 
	line  36:  "SD1_DATA0"       unused   input  active-high 
	line  37:  "SD1_DATA1"       unused   input  active-high 
	line  38:  "SD1_DATA2"       unused   input  active-high 
        line  39:  "SD1_DATA3"       unused   input  active-high 
	line  40:   "PWM0_OUT"       unused   input  active-high 
	line  41:   "PWM1_OUT"       unused   input  active-high 
	line  42:    "ETH_CLK"       unused   input  active-high 
	line  43:   "WIFI_CLK"       unused   input  active-high 
	line  44:       "SDA0"       unused   input  active-high 
	line  45:       "SCL0"       unused   input  active-high 
	line  46:   "SMPS_SCL"       unused   input  active-high 
	line  47:   "SMPS_SDA"       unused  output  active-high 
	line  48:   "SD_CLK_R"       unused   input  active-high 
	line  49:   "SD_CMD_R"       unused   input  active-high 
	line  50: "SD_DATA0_R"       unused   input  active-high 
	line  51: "SD_DATA1_R"       unused   input  active-high 
	line  52: "SD_DATA2_R"       unused   input  active-high 
	line  53: "SD_DATA3_R"       unused   input  active-high 
gpiochip1 - 8 lines:
	line   0:      "BT_ON"       unused  output  active-high 
	line   1:      "WL_ON"       unused  output  active-high 
	line   2:  "PWR_LED_R"        "PWR"  output   active-low [used]
	line   3:    "LAN_RUN"       unused  output  active-high 
	line   4:         "NC"       unused   input  active-high 
	line   5:  "CAM_GPIO0" "cam1_regulator" output active-high [used]
	line   6:  "CAM_GPIO1"       unused  output  active-high 
	line   7:         "NC"       unused   input  active-high

Matzeonom, 2024-04-06:
So, and hopefully finally: Thank you very much for the crazy quick support !

Tom-Hirschberger, 2024-04-06:
Great to hear that. 
The problem was caused by the library not knowing anything about the Raspberry 3B Plus (but the Raspberry 3B) and a mistake in my implementation which uses the gpioinfo.json in that case.

A other user had a different problem but was using the same model which pointed me to the problem

closed 2024-04-06 by Tom-Hirschberger