Tom-Hirschberger/MMM-GPIO-Notifications #28: Since 0.2.2 no more monitor activation

det-box, 2024-04-06:
Hello,
Since the update to 0.2.2, movement is no longer detected.
As a result, of course, the monitor is not activated.
I use Raspi4 with bullsey and MM 2.27.0.
`Linux mirror2 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
After copying node_helper.js, from version 0.2.1, into the MMM-GPIO-. folder
the activation of the monitor works and the update message is gone.

No message in the LOG for movement in front of the PIR.
```
|mm  | [2024-04-06 10:10:15.368] [LOG]   [CALDAV] File: ICLOUD_Privat.ics is refreshed. 
0|mm  | [2024-04-06 10:10:54.442] [LOG]   MMM-Screen-Powersave-Notification: Disabled screen timeout! 
```
No message in the LOG for movement in front of the PIR.
`0|mm  | [2024-04-06 10:22:03.511] [LOG]   MMM-GPIO-Notifications: Watched pin: 22 triggered with value 0! 
0|mm  | [2024-04-06 10:22:03.512] [LOG]   MMM-GPIO-Notifications: Sending notifications of pin 22... 
0|mm  | [2024-04-06 10:22:03.513] [LOG]   MMM-GPIO-Notifications: Skipped notifications of pin 22 cause the state 0 has no notifications configured. 
0|mm  | [2024-04-06 10:22:52.424] [LOG]   MMM-Screen-Powersave-Notification: Turning screen off! 
0|mm  | [2024-04-06 10:22:52.559] [LOG]   MMM-Screen-Powersave-Notification: Running all scripts in: /home/pi/MagicMirror/modules/MMM-Screen-Powersave-Notification/callbackScripts/off 
0|mm  | [2024-04-06 10:22:52.561] [LOG]   MMM-Screen-Powersave-Notification: Disabled screen timeout! 
0|mm  | [2024-04-06 10:22:52.567] [LOG]   MMM-Screen-Powersave-Notification:   halloWorld.bash `

Thanks

Tom-Hirschberger, 2024-04-06:
Hi and sorry.

Can you please provide some more information.
* Which board do you use?
* Which operating system do you use?
* Can you please share your module configuration?
* Can you please post all Log messages of the GPIO module? Especially the ones about the pin registration?
* Are there any errors in the error log file?

det-box, 2024-04-06:
My config:
`		{	
    		    module: 'MMM-GPIO-Notifications',
    		    config: {
        		'22': {	//PIR Eingang
				delay: 10000,
          			notifications_high: [
            			  {
              			    notification: 'USER_PRESENCE',
              			    payload: true
            			  },
            			  {
              			    notification: 'SCREEN_ON',
              			    payload: { 'forced': false }
            			  }
              			]
        		    }
    		    }
		},
`
Board and System:
Raspberry4
`root@mirror2:/home/pi/.pm2/logs# lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
root@mirror2:/home/pi/.pm2/logs# `
`root@mirror2:/home/pi/.pm2/logs# uname -a
Linux mirror2 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`

here is the log of GPIO registration
`[2024-04-06 11:02:58.021] [LOG]   Connecting socket for: MMM-GPIO-Notifications 
[2024-04-06 11:02:58.026] [LOG]   Could not find any device information for RaspberryPi_3BPlus. Using information of gpioinfo.json! 
`
hope this helps
thanks

Tom-Hirschberger, 2024-04-06:
Ok.

The library does not have any information about Rasbperry 3B Plus yet thats why it uses the information of the gpioinfo.json file created during the module installation.

What does your `gpioinfo.json` look like and is there any more information in the log about the registration of the pins?
Something like  "Using chip XX and line XX info of 22"?

Can you provide the output of the `gpioinfo` command please?

det-box, 2024-04-06:
heer the Info:
pi@mirror2:~/MagicMirror/modules/MMM-GPIO-Notifications $ gpioinfo
gpiochip0 - 54 lines:
	line   0:     "ID_SDA"       unused   input  active-high 
	line   1:     "ID_SCL"       unused   input  active-high 
	line   2:       "SDA1"       unused   input  active-high 
	line   3:       "SCL1"       unused   input  active-high 
	line   4:  "GPIO_GCLK"       unused   input  active-high 
	line   5:      "GPIO5"       unused   input  active-high 
	line   6:      "GPIO6"       unused   input  active-high 
	line   7:  "SPI_CE1_N"       unused   input  active-high 
	line   8:  "SPI_CE0_N"       unused   input  active-high 
	line   9:   "SPI_MISO"       unused   input  active-high 
	line  10:   "SPI_MOSI"       unused   input  active-high 
	line  11:   "SPI_SCLK"       unused   input  active-high 
	line  12:     "GPIO12"       unused   input  active-high 
	line  13:     "GPIO13"       unused   input  active-high 
	line  14:       "TXD1"       unused   input  active-high 
	line  15:       "RXD1"       unused   input  active-high 
	line  16:     "GPIO16"       unused   input  active-high 
	line  17:     "GPIO17"       unused   input  active-high 
	line  18:     "GPIO18"       unused   input  active-high 
	line  19:     "GPIO19"      "sysfs"   input  active-high [used]
	line  20:     "GPIO20"      "sysfs"   input  active-high [used]
	line  21:     "GPIO21"       unused   input  active-high 
	line  22:     "GPIO22" "opengpio_0_22_watch" input active-high [used]
	line  23:     "GPIO23"       unused   input  active-high 
	line  24:     "GPIO24"       unused   input  active-high 
	line  25:     "GPIO25"       unused   input  active-high 
	line  26:     "GPIO26"      "sysfs"   input  active-high [used]
	line  27:     "GPIO27"       unused   input  active-high 
	line  28: "HDMI_HPD_N"       unused   input  active-high 
	line  29: "STATUS_LED_G" "ACT" output active-high [used]
	line  30:       "CTS0"       unused   input  active-high 
	line  31:       "RTS0"       unused   input  active-high 
	line  32:       "TXD0"       unused   input  active-high 
	line  33:       "RXD0"       unused   input  active-high 
	line  34:    "SD1_CLK"       unused   input  active-high 
	line  35:    "SD1_CMD"       unused   input  active-high 
	line  36:  "SD1_DATA0"       unused   input  active-high 
	line  37:  "SD1_DATA1"       unused   input  active-high 
	line  38:  "SD1_DATA2"       unused   input  active-high 
	line  39:  "SD1_DATA3"       unused   input  active-high 
	line  40:   "PWM0_OUT"       unused   input  active-high 
	line  41:   "PWM1_OUT"       unused   input  active-high 
	line  42:    "ETH_CLK"       unused   input  active-high 
	line  43:   "WIFI_CLK"       unused   input  active-high 
	line  44:       "SDA0"       unused   input  active-high 
	line  45:       "SCL0"       unused   input  active-high 
	line  46:   "SMPS_SCL"       unused   input  active-high 
	line  47:   "SMPS_SDA"       unused  output  active-high 
	line  48:   "SD_CLK_R"       unused   input  active-high 
	line  49:   "SD_CMD_R"       unused   input  active-high 
	line  50: "SD_DATA0_R"       unused   input  active-high 
	line  51: "SD_DATA1_R"       unused   input  active-high 
	line  52: "SD_DATA2_R"       unused   input  active-high 
	line  53: "SD_DATA3_R"       unused   input  active-high 
gpiochip1 - 8 lines:
	line   0:      "BT_ON"       unused  output  active-high 
	line   1:      "WL_ON"       unused  output  active-high 
	line   2:  "PWR_LED_R"        "PWR"  output  active-high [used]
	line   3:    "LAN_RUN"       unused  output  active-high 
	line   4:         "NC"       unused   input  active-high 
	line   5:  "CAM_GPIO0" "cam1_regulator" output active-high [used]
	line   6:  "CAM_GPIO1"       unused  output  active-high 
	line   7:         "NC"       unused   input  active-high

det-box, 2024-04-06:
[gpioinfo.json](https://github.com/Tom-Hirschberger/MMM-GPIO-Notifications/files/14893100/gpioinfo.json)

leider keine weiteren Meldungen im LOG in der Art
"Using chip XX and line XX info of 22"

Tom-Hirschberger, 2024-04-06:
Anything in the error-log (usually `/home/pi/.pm2/logs/MagicMirror-error.log`)

det-box, 2024-04-06:
mm-error.log
[2024-04-06 11:37:53.937] [ERROR] ERROR! Could not find main module js file for calendars 
[2024-04-06 11:38:10.969] [WARN]  MMM-Fuel: Filtered stationIds list is empty 
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT
[2024-04-06 11:39:17.500] [ERROR] ERROR! Could not find main module js file for calendars 
[2024-04-06 11:39:34.807] [WARN]  MMM-Fuel: Filtered stationIds list is empty 

MagicMirror-error.log
[2997:0427/151053.270763:ERROR:viz_main_impl.cc(188)] Exiting GPU process due to errors during initialization
[3039:0427/151053.541260:ERROR:sandbox_linux.cc(377)] InitializeSandbox() called with multiple threads in process gpu-process.
/home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT

Tom-Hirschberger, 2024-04-06:
The error log looks fine to me. 
Looks like the module tries to watch GPIO2 instead of GPIO22. I will try to figure this out

Tom-Hirschberger, 2024-04-06:
Can you provide the output of `cat /proc/device-tree/model` to me in the meantime please?
I need the info to add the board to the opengpio library soon

det-box, 2024-04-06:
pi@mirror2:~/MagicMirror/modules/MMM-GPIO-Notifications $ cat /proc/device-tree/model
Raspberry Pi 3 Model B Plus Rev 1.3
pi@mirror2:~/MagicMirror/modules/MMM-GPIO-Notifications $

Tom-Hirschberger, 2024-04-06:
Think i found the problem. Can you please try version 0.2.5 released a few seconds ago

det-box, 2024-04-06:
with 0.2.5
The monitor is now reactivated
LOG
`
0|mm  | [2024-04-06 12:18:57.248] [LOG]   Connecting socket for: MMM-GPIO-Notifications 
0|mm  | [2024-04-06 12:18:57.252] [LOG]   Could not find any device information for RaspberryPi_3BPlus. Using information of gpioinfo.json! 
0|mm  | [2024-04-06 12:18:57.257] [LOG]   Connecting socket for: MMM-Screen-Powersave-Notification 
`
and 
`|mm  | [2024-04-06 12:20:19.444] [LOG]   MMM-Screen-Powersave-Notification: Disabled screen timeout! 
0|mm  | [2024-04-06 12:20:42.135] [LOG]   Rising Event true 
0|mm  | [2024-04-06 12:20:42.136] [LOG]   MMM-GPIO-Notifications: Watched pin: 22 triggered with value 1! 
0|mm  | [2024-04-06 12:20:42.139] [LOG]   MMM-GPIO-Notifications: Sending notifications of pin 22... 
0|mm  | [2024-04-06 12:20:42.140] [LOG]   MMM-GPIO-Notifications: Sending notifications for high state of pin 22... 
0|mm  | [2024-04-06 12:20:42.224] [LOG]   MMM-Screen-Powersave-Notification: Turning screen on! 
0|mm  | [2024-04-06 12:20:43.894] [LOG]   MMM-Screen-Powersave-Notification: Running all scripts in: /home/pi/MagicMirror/modules/MMM-Screen-Powersave-Notification/callbackScripts/on 
0|mm  | [2024-04-06 12:20:44.027] [LOG]   MMM-Screen-Powersave-Notification: Resetted screen timeout to 60 seconds! 
0|mm  | [2024-04-06 12:20:44.134] [LOG]   MMM-Screen-Powersave-Notification: Resetted screen timeout to 60 seconds! 
0|mm  | [2024-04-06 12:20:44.141] [LOG]   MMM-Screen-Powersave-Notification:   halloWorld.bash 
0|mm  | [2024-04-06 12:20:44.646] [LOG]   Falling Event false 
0|mm  | [2024-04-06 12:20:44.647] [LOG]   MMM-GPIO-Notifications: Watched pin: 22 triggered with value 0! 
0|mm  | [2024-04-06 12:20:44.649] [LOG]   MMM-GPIO-Notifications: Sending notifications of pin 22... 
0|mm  | [2024-04-06 12:20:44.652] [LOG]   MMM-GPIO-Notifications: Skipped notifications of pin 22 cause the state 0 has no notifications configured. 
0|mm  | [2024-04-06 12:20:47.788] [LOG]   Rising Event true 
0|mm  | [2024-04-06 12:20:47.789] [LOG]   MMM-GPIO-Notifications: Watched pin: 22 triggered with value 1! 
0|mm  | [2024-04-06 12:20:47.792] [LOG]   MMM-GPIO-Notifications: Sending notifications of pin 22... 
0|mm  | [2024-04-06 12:20:47.795] [LOG]   MMM-GPIO-Notifications: Skipping pin 22 because the delay is not exceeded ! 
0|mm  | [2024-04-06 12:21:03.066] [LOG]   Falling Event false 
0|mm  | [2024-04-06 12:21:03.067] [LOG]   MMM-GPIO-Notifications: Watched pin: 22 triggered with value 0! 
0|mm  | [2024-04-06 12:21:03.070] [LOG]   MMM-GPIO-Notifications: Sending notifications of pin 22... 
0|mm  | [2024-04-06 12:21:03.073] [LOG]   MMM-GPIO-Notifications: Skipped notifications of pin 22 cause the state 0 has no notifications configured. 
^C`
gpioinfo:
`pi@mirror2:~/MagicMirror/modules/MMM-GPIO-Notifications $ gpioinfo
gpiochip0 - 54 lines:
	line   0:     "ID_SDA"       unused   input  active-high 
	line   1:     "ID_SCL"       unused   input  active-high 
	line   2:       "SDA1"       unused   input  active-high 
	line   3:       "SCL1"       unused   input  active-high 
	line   4:  "GPIO_GCLK"       unused   input  active-high 
	line   5:      "GPIO5"       unused   input  active-high 
	line   6:      "GPIO6"       unused   input  active-high 
	line   7:  "SPI_CE1_N"       unused   input  active-high 
	line   8:  "SPI_CE0_N"       unused   input  active-high 
	line   9:   "SPI_MISO"       unused   input  active-high 
	line  10:   "SPI_MOSI"       unused   input  active-high 
	line  11:   "SPI_SCLK"       unused   input  active-high 
	line  12:     "GPIO12"       unused   input  active-high 
	line  13:     "GPIO13"       unused   input  active-high 
	line  14:       "TXD1"       unused   input  active-high 
	line  15:       "RXD1"       unused   input  active-high 
	line  16:     "GPIO16"       unused   input  active-high 
	line  17:     "GPIO17"       unused   input  active-high 
	line  18:     "GPIO18"       unused   input  active-high 
	line  19:     "GPIO19"      "sysfs"   input  active-high [used]
	line  20:     "GPIO20"      "sysfs"   input  active-high [used]
	line  21:     "GPIO21"       unused   input  active-high 
	line  22:     "GPIO22" "opengpio_0_22_watch" input active-high [used]
	line  23:     "GPIO23"       unused   input  active-high 
	line  24:     "GPIO24"       unused   input  active-high 
	line  25:     "GPIO25"       unused   input  active-high 
	line  26:     "GPIO26"      "sysfs"   input  active-high [used]
	line  27:     "GPIO27"       unused   input  active-high 
	line  28: "HDMI_HPD_N"       unused   input  active-high 
	line  29: "STATUS_LED_G" "ACT" output active-high [used]
	line  30:       "CTS0"       unused   input  active-high 
	line  31:       "RTS0"       unused   input  active-high 
	line  32:       "TXD0"       unused   input  active-high 
	line  33:       "RXD0"       unused   input  active-high 
	line  34:    "SD1_CLK"       unused   input  active-high 
[... 6 lines cut ...]
	line  41:   "PWM1_OUT"       unused   input  active-high 
	line  42:    "ETH_CLK"       unused   input  active-high 
	line  43:   "WIFI_CLK"       unused   input  active-high 
	line  44:       "SDA0"       unused   input  active-high 
	line  45:       "SCL0"       unused   input  active-high 
	line  46:   "SMPS_SCL"       unused   input  active-high 
	line  47:   "SMPS_SDA"       unused  output  active-high 
	line  48:   "SD_CLK_R"       unused   input  active-high 
	line  49:   "SD_CMD_R"       unused   input  active-high 
[... cut ...]
	line   1:      "WL_ON"       unused  output  active-high 
	line   2:  "PWR_LED_R"        "PWR"  output  active-high [used]
	line   3:    "LAN_RUN"       unused  output  active-high 
	line   4:         "NC"       unused   input  active-high 
	line   5:  "CAM_GPIO0" "cam1_regulator" output active-high [used]
	line   6:  "CAM_GPIO1"       unused  output  active-high 
	line   7:         "NC"       unused   input  active-high 
pi@mirror2:~/MagicMirror/modules/MMM-GPIO-Notifications $ 
`
Thank you very much, I think it's solved

closed 2024-04-06 by det-box