Tom-Hirschberger/MMM-GPIO-Notifications #30: gpioinfo empty

mholzma, 2024-04-16:
I am running MagicMirror in a docker container using the :fat tag and had everything working fine. Every once in awhile, when I reboot, I have to go back in and npm install the MMM-GPIO-Notifications folder again or I just have a blank screen.

This last time, after using the "docker exec -it mm bash" command, I had a lot of trouble getting the postinstall to work. I ended up wiping the directory and reinstalling which left me a different version with no ./postinstall.

running the gpioinfo was returning an empty string unless I run docker-compose in privileged mode. The problem is running in privileged causes a repeating crash with stderr and polling that I haven't figured out. Instead I run the container with the mapping of:
    devices:
      - /dev/vchiq
      - /dev/gpiomem

Not sure how to resolve this on my Raspberry Pi3

Tom-Hirschberger, 2024-04-16:
Hi,

as of the new release of the module i changed to a different library to access the GPIO pins (opengpio) instead of onoff which used the deprecated sysinfo interface. Had a lot of trouble with the onoff lib with the latest 6.6 kernel.
The new lib does not need the postinstall script but needs the preinstall script to be run instead.
Just tested with the latest karsten13/magicmirrir:fat image and everything works on my Pi3 but not without the privileged option.
Maybe you can try to map the gpio device to the container instead of the whole /dev.

You can run gpioinfo directly on the device and check which is the gpio chip to use. You then will find a corresponding /dev/gpiochipXX device and can map it to the container.

mholzma, 2024-04-16:
Thanks,

I just tried mapping /dev/gpiochip0 and /dev/gpiochip1 and /dev/gpiomem after running gpiodetect and gpioinfo outside the container.

Unfortunately, it still seems to be having issues
[2024-04-16 15:12:26.123] [LOG]   Connecting socket for: MMM-GPIO-Notifications
[2024-04-16 15:12:26.125] [LOG]   Could not find any device information. Using information of gpioinfo.json!
[2024-04-16 15:12:32.373] [LOG]   MMM-GPIO-Notifications: Trying to registering pin: 11
[2024-04-16 15:12:32.373] [LOG]   MMM-GPIO-Notifications: Failed to register pin: 11 cause we could not find any information about the gpio chip and lane to use!

Maybe there is a permission problem I need to solve. Also thinking about rebuilding my entire Pi3 to try to fix whatever the privilege problem is.

Tom-Hirschberger, 2024-04-16:
Hi,

just tried the same but without luck.
It looks like the kernel blocks any access to the GPIO pins if the container does not run in privilege mode.
Could not find any examples without privileged mode.

mholzma, 2024-04-16:
Thanks.. makes sense and I'll try to figure out why privileged mode isn't working for me.

closed 2024-04-16 by mholzma

mholzma, 2024-04-18:
Turns out that the MMM-GPIO-Notification module is causing my system crash in privileged mode. My assumption is that there is something in the new library or some other conflict causing MagicMirror to crash when the GPIO is triggered.

Tom-Hirschberger, 2024-04-18:
I will try to investigate this further but I will need some time.

The developer of the opengpio lib implemented some error handling for me in the latest version.
I will check if we have overseen some cases.

Tom-Hirschberger, 2024-04-21:
Looks like the container images are missing the dependencies at the moment. Wrote a message to karsten13 but he has not responded yet.
Tried to build a custom image based on his fat one the the dependencies added. Everything works well then ( in privilege model).
Will wait for the answer of karsten13 and update the instructions

mholzma, 2024-04-21:
Thanks again. I ended up rebuilding my RaspberryPi and since it is the only thing running, went native instead of the docker container. Works great again.